RoboticsBrno/RB0002-BatteryPack #10: Wrong KiKit panelize syntax in makefile

bsaid, 2022-04-20:
```
/RB0002-BatteryPack$ make
kikit panelize extractboard -s 175 50 70 40 batteryPack/batteryPack.kicad_pcb build/charger.kicad_pcb
Usage: kikit panelize [OPTIONS] INPUT OUTPUT
Try 'kikit panelize --help' for help.

Error: Got unexpected extra arguments (70 40 batteryPack/batteryPack.kicad_pcb build/charger.kicad_pcb)
make: *** [Makefile:12: build/charger.kicad_pcb] Error 2
```

Platform: Xubuntu 20.04, Python 3.8.10, KiKit version 1.0.5

Maybe the syntax changed with the new version of KiKit?
This project is mentioned as an example project inside KiKit README, so this is why I care about this issue.

yaqwsx, 2022-04-20:
Yes, this project uses KiKit v0.4, however, there were major breaking changes in the v1.0 release.

Where did you find the mention of a Battery pack being an example? All the notes about it should be gone.

bsaid, 2022-04-20:
[https://github.com/yaqwsx/KiKit/blob/master/README.md](https://github.com/yaqwsx/KiKit/blob/master/README.md?plain=1) line 65

yaqwsx, 2022-04-20:
Thanks, I removed the notes. I think you might want to start with: https://github.com/yaqwsx/KiKit/blob/master/doc/examples.md

bsaid, 2022-04-20:
I started there, but I was looking for how to make a panel from several different PCBs. Can you add this case to your examples?

closed 2022-04-20 by bsaid

yaqwsx, 2022-04-20:
That's a duplicate of:
- https://github.com/yaqwsx/KiKit/issues/283
- https://github.com/yaqwsx/KiKit/discussions/327

Currently, this is somewhere on my Todo list, but low in priority.